30350n/inventree-part-import #3: IndexError: list index out of range

si-hb, 2024-01-08:
Appears to be caused by TME not returning any data for a part, therefore index [0] doesn't exit? I don't get the error if the TME part exists.

`$ inventree-part-import DRV8837DSGR`

searching for DRV8837DSGR ...
Traceback (most recent call last):

....

packages/inventree_part_import/suppliers/supplier_tme.py", line 23, in search
    tme_stock = self.tme_api.get_prices_and_stocks([tme_part["Symbol"]])[0]
IndexError: list index out of range```

30350n, 2024-01-08:
Hm do you always get that error when trying to import that specific part?
> I don't get the error if the TME part exists.

`DRV8837DSGR` exists at tme though. It also just works fine for me.

30350n, 2024-01-08:
Are you using an anonymous or private TME API token?

si-hb, 2024-01-09:
I believe private? I got it from [here](https://developers.tme.eu/dev): 
[image]

si-hb, 2024-01-09:
> Hm do you always get that error when trying to import that specific part?

Tested again and appears so, yes.

30350n, 2024-01-09:
Oh, I got an idea! What's your language and location set to?

si-hb, 2024-01-10:
> Oh, I got an idea! What's your language and location set to?

EN/CA

30350n, 2024-01-10:
Do you have your currency set to CAD by any chance?

si-hb, 2024-01-10:
> Do you have your currency set to CAD by any chance?

Yes

30350n, 2024-01-10:
Ah! That's the problem! TME only supports EUR, USD and GBP. If you still want pricing information for tme you can overwrite the currency for TME only by specifying `currency: USD` (for example) under the `TME` section in your `suppliers.yaml`.

30350n, 2024-01-10:
I'll add a warning + actual fix for the error tomorrow.

30350n, 2024-01-10:
Got a simple fix for the error for now, but I'd really like to have a warning/interactive resolving as well, but that's going to require some more refactoring ...

30350n, 2024-02-24:
You'll get an error now, but no interactive resolving. Going to leave this open, because I'll probably still have to do the refactoring eventually ...