30350n/inventree-part-import #4: TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

si-hb, 2024-01-08:
`$ inventree-part-import NUD3124LT1GOSTR-ND`

```(venv) hb@HB-Ryzen:~/Documents/python/Inventree$ inventree-part-import NUD3124LT1GOSTR-ND
(only 3 of 5 available supplier modules are configured)
warning: category 'Electronics - uncategorized' on host is not defined in categories.yaml
warning: category 'Furniture' on host is not defined in categories.yaml
successfully setup categories!

searching for NUD3124LT1GOSTR-ND ...
(no results at TME)
(no results at Mouser)
Traceback (most recent call last):4LT1GOSTR-ND ...
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree/base.py", line 51, in __init__
    pk = int(pk)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/.local/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
  File "/home/h/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/cli.py", line 21, in wrapper
    func(*args, **kwargs)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/cli.py", line 171, in inventree_part_import
    last_import_result = importer.import_part(part, supplier, only_supplier)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 76, in import_part
    import_result |= self.import_supplier_part(supplier, api_part)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 116, in import_supplier_part
    manufacturer_part = ManufacturerPart(self.api, supplier_part.manufacturer_part)
  File "/home/hb/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree/base.py", line 53, in __init__
    raise TypeError(f"Supplied <pk> value ({pk}) for {self.__class__} is invalid.")
TypeError: Supplied <pk> value (None) for <class 'inventree.company.ManufacturerPart'> is invalid.```

30350n, 2024-01-09:
This is an edge case I didn't think was possible. Should be fixed on master, would be great if you could give it a try ^^

si-hb, 2024-01-09:
Test working, though it should be noted that I've since ensured my DB has a manufacturer part for all parts, which was not the case at the time of the error.

Closing issue at any rate...

closed 2024-01-09 by si-hb