30350n/inventree-part-import #11: Escape '%' in TME URL

ondras12345, 2024-02-24:
Hi,
I've just discovered this repository. Thank you for your great work!

When importing this part:
```
inventree-part-import -o tme SMD1206-10K-1%
```
I get a broken link in the manufacturer part: https://www.tme.eu/en/details/smd1206-10k-1%/smd-resistors/royal-ohm/1206s4f1002t5e/
The correct URL is https://www.tme.eu/en/details/smd1206-10k-1%25/smd-resistors/royal-ohm/1206s4f1002t5e/

30350n, 2024-02-24:
Thanks for reporting! This should be fixed on master now.

You can install from master via `pipx install git+https://github.com/30350n/inventree_part_import.git`, but make sure to uninstall other versions first.

30350n, 2024-02-25:
Oh yeah, that was one to many ^^
Fixed!

ondras12345, 2024-02-25:
Something is still not quite right.
```console
$ inventree-part-import -o tme SMD1206-10K-1%
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for SMD1206-10K-1% ...
updating TME part ...
error: failed to import part with: 
    link: ['Enter a valid URL.']

the following parts failed to import:
SMD1206-10K-1%

$ inventree-part-import -o tme ESD5Z3.3T1G
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for ESD5Z3.3T1G ...
updating TME part ...
error: failed to import part with: 
    link: ['Enter a valid URL.']

the following parts failed to import:
ESD5Z3.3T1G                                    
```

With verbose:
```console
$ inventree-part-import --verbose -o tme SMD1206-10K-1%
setting up InvenTree API ...
loading api configuration from 'inventree.yaml' ...
(only 1 of 5 available supplier modules are configured)
setting up categories ...
successfully setup categories!

searching for SMD1206-10K-1% ...
searching at TME ...
found existing TME part SMD1206-10K-1% ...
updating TME part ...
error: failed to import part with: 
    link: ['Enter a valid URL.']

FULL TRACEBACK:
Traceback (most recent call last):
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 85, in import_part
    import_result |= self.import_supplier_part(supplier, api_part, existing_part)
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 203, in import_supplier_part
    update_object_data(supplier_part, supplier_part_data, f"{supplier.name} part")
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/inventree_helpers.py", line 77, in update_object_data
    obj.save(data)
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree/base.py", line 242, in save
    response = self._api.patch(self._url, data, files=files)
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree/api.py", line 464, in patch
    response = self.request(
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/retries.py", line 68, in request
    return super().request(api_url, **kwargs)
  File "/home/ondra/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree/api.py", line 378, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'http://inventree.rpi.lab.lan/api/company/part/3/', 'method': 'PATCH', 'status_code': 400, 'body': '{"link":["Enter a valid URL."]}', 'headers': {'AUTHORIZATION': '[redacted]'}, 'params': {'format': 'json'}, 'data': {'part': 3, 'manufacturer_part': 3, 'supplier': 3, 'SKU': 'SMD1206-10K-1%', 'description': 'Resistor: thick film; SMD; 1206; 10kΩ; 0.25W; ±1%; -55÷125°C', 'link': 'https%3A//www.tme.eu/en/details/smd1206-10k-1%25/smd-resistors/royal-ohm/1206s4f1002t5e/', 'packaging': '', 'available': 660600.0}}

the following parts failed to import:
SMD1206-10K-1%
```

30350n, 2024-02-25:
Ah I see it's escaping the colon ... (Sorry for not catching these, I was testing in dry mode).
Pushed another fix.

ondras12345, 2024-02-25:
Thanks, now it's working correctly.

closed 2024-02-25 by ondras12345