30350n/inventree-part-import #12: TME API does not support 3 character ISO3166 location codes

ondras12345, 2024-02-24:
I had this in my `config.yaml`:
```yaml
location: CZE
```
The initial interactive setup process let me input this value (it passed validation).
However, when trying to import a part from TME, it fails with no useful error message:
```console
$ inventree-part-import -o tme SMD1206-10K-1%
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for SMD1206-10K-1% ...
(no results at TME)

the following parts failed to import:
SMD1206-10K-1%
```
After changing the config to
```yaml
location: CZ
```
It imports fine.

30350n, 2024-02-24:
This one should also be fixed, I also added some additional checks for DigiKey and Reichelt.

ondras12345, 2024-02-25:
It still says `(no results at TME)` when my location is set to `CZE`.

I did not test DigiKey nor Reichelt as I am not currently using those.

30350n, 2024-02-25:
Okay, that one is obvious, I did all the checking for 2/3 letter codes and everything but didn't actually use the results 🤦 .
#15 is more annoying though, because that's just TME using invalid languages codes. I've reorganized the code now a bit so both of these should also be fixed.

ondras12345, 2024-02-25:
Thanks, this one is fixed.

closed 2024-02-25 by ondras12345