30350n/inventree-part-import #13: warning: datasheet [...] has invalid file extension 'PDF'

ondras12345, 2024-02-24:
I believe the message is coming from here:
https://github.com/30350n/inventree_part_import/blob/0eeb99a91d55c5003e3960c58991fe1a287126ae/inventree_part_import/inventree_helpers.py#L147C26-L149

```console
$ inventree-part-import -o tme ESD5Z3.3T1G
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for ESD5Z3.3T1G ...
warning: datasheet 'https://www.tme.eu/Document/71395020f2f7b5f6cf374e0fb9bce274/ESD5Zx-DTE.PDF' has invalid file extension 'PDF'
successfully updated TME part ESD5Z3.3T1G ([redacted])

successfully imported all parts!
```

30350n, 2024-02-24:
Yep exactly, good catch. Also fixed!

ondras12345, 2024-02-25:
Thanks, now it is working.

closed 2024-02-25 by ondras12345