30350n/inventree-part-import #14: Fix DigiKey search, Update to API v4

30350n, 2024-02-25:
DigiKey search is somewhat broken right now, the following things need fixing:
- [ ] DigiKey Part Number search doesn't work
- [ ] Parts with multiple packages only return pricing information for one of them

The [`digikey-api`]() also seems like it's been unmaintained for quite a while now and doesn't support the new [Product Information V4](https://developer.digikey.com/products/product-information-v4) endpoint, which could be helpful here.

sle118, 2024-03-07:
Have you had a chance to evaluate the effort? Would you fork the digikey-api altogether, or just implement your own subset of APIs? I might be interested to chip in, although Python isn't my primary language.

30350n, 2024-03-07:
I've started to twiddle around with it a bit, but haven't really gotten anywhere. The perfectionist in me wants to do the whole API generation in a similar way as it's currently done in [`digikey-api` here](https://github.com/peeter123/digikey-api/blob/bf24772fa55d6ed457bc5c862a9165985cff10a8/generate-v3-api/generage_client_swagger.py), but ideally while building the actual package (so that none of the generated code would actually be part of the git repository).

Things I've learned:
- [`swagger-to`](https://pypi.org/project/swagger-to/) is poorly implemented and doesn't work with DigiKey swagger files
- [`swagger-codegen-cli`](https://github.com/swagger-api/swagger-codegen) (this is what `digikey-api` uses) works, but generates a lot of bloat that has to be sorted through

30350n, 2024-06-30:
Searching for DigiKey part numbers should work again now in [1.7](https://github.com/30350n/inventree_part_import/releases/tag/1.7).

30350n, 2025-10-04:
I've dropped the dependency on [`digikey-api`](https://github.com/peeter123/digikey-api) and implemented the relevant API calls of the v4 API now in [`1c920eb`](https://github.com/30350n/inventree-part-import/commit/1c920eb85295915d1effaaf9671d5d38164e1759). This should atleast get DigiKey as a supplier to a useable state again.

Here's a few things that would be nice to have, but haven't been implemented yet:
- [ ] If searching for a MPN, import **all** relevant Supplier Parts (DigiKey uses a SKU per packaging variant)
- [ ] Support `MyPricing`
- [ ] Support `X-DIGIKEY-Customer-Id`

30350n, 2025-10-04:
(Would be great if people could give this a spin to see if there's any issues. I'm planning to push out a new release later next week.)