30350n/inventree-part-import #15: Incorrect TME links for `language: CS`

ondras12345, 2024-02-25:
When setting this in my `config.yaml`:
```yaml
language: CS
```
Links for TME parts are generated like this: https://www.tme.eu/cs/details/esd5z3.3t1g/transil-diody-jednosmerne-smd/onsemi/
Whereas the correct link is: https://www.tme.eu/cz/details/esd5z3.3t1g/transil-diody-jednosmerne-smd/onsemi/

Previously, I worked around this by setting
```yaml
language: CZ
```
which is not a valid ISO639 code. The new version (as of f6e8cf9) correctly complains about this:
```console
$ inventree-part-import -o tme ESD5Z3.3T1G
error: failed to load 'TME' supplier module (invalid language code 'CZ')
Usage: inventree-part-import [OPTIONS] [INPUTS]...
Try 'inventree-part-import --help' for help.

Error: Invalid value for '-o' / '--only': 'tme' is not one of .
```
However, this means that I can no longer get a working link.

30350n, 2024-02-25:
As mentioned in #12, TME seems to be the issue here. I simply accept "invalid" language codes now, if the TME api supports them. So you should set `language: cz` for TME.

30350n, 2024-02-25:
You can override this setting **only** for TME btw, by defining it in the `TME` section in `suppliers.yaml`.

ondras12345, 2024-02-25:
It still says
```console
$ inventree-part-import -o tme ESD5Z3.3T1G
error: failed to load 'TME' supplier module (invalid language code 'cz')
Usage: inventree-part-import [OPTIONS] [INPUTS]...
Try 'inventree-part-import --help' for help.

Error: Invalid value for '-o' / '--only': 'tme' is not one of .
```
No matter whether I set `language: CZ` in `config.yaml` or `suppliers.yaml`.

30350n, 2024-02-25:
Ah great so this is an **internal** TME bug ...
Their `Utils/GetCountries` API endpoint returns:
```
['ar', 'bg', 'cs', 'de', 'el', 'en', 'es', 'et', 'fi', 'fr', 'hr', 'hu', 'it', 'ja', 'ko', 'lt', 'lv', 'nl', 'pl', 'pt', 'ro', 'ru', 'sk', 'sl', 'sv', 'tr', 'uk', 'vi', 'zh']
```

I've added a url check for `tme.eu/cs/` now and I am just replacing that with `tme.eu/cz/`. Ugly, but should hopefully do the trick.

30350n, 2024-02-25:
So `language: cs` should work now after all.

ondras12345, 2024-02-25:
Thanks, working with `language: cs`.

closed 2024-02-25 by ondras12345