30350n/inventree-part-import #16: When importing multiple parts, it asks for the same category mapping multiple times

ondras12345, 2024-02-27:
Let's say I want to import two connectors from TME.
`categories.yaml` before:
```yaml
            RJ connectors:
                _description: RJ Modular Connectors
                _parameters:
                    - Connector Type
                    - Ethernet Category
```

```console
$ inventree-part-import -o tme 54601-908WPLF KEYS948
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for 54601-908WPLF ...

failed to match category for 'Konektory / Konektory pro přenos dat / Konektory RJ', select category:
  Electronics / Connectors
  Electronics / Connectors / Jumpers
  Electronics / Connectors / Audio Connectors
  Electronics / Connectors / Barrel Connectors
  Electronics / Connectors / Circular Connectors
> Enter Manually ...
  Skip ...
category name: RJ Connectors

failed to match some parameters from 'https://www.tme.eu/cz/details/54601-908wplf/konektory-rj/amphenol-communications-solutions/'

failed to match value for parameter 'Ethernet Category', select value:
> Cat 3                             | Kategorie Ethernet
  na PCB                            | Mechanická montáž
  zásuvka                           | Konektor
  Amphenol Communications Solutions | Výrobce
  úhlové 90°                        | Prostorová orientace
  Enter Value Manually ...
  Skip ...
successfully added TME part 54601-908WPLF (http://inventree.rpi.lab.lan/supplier-part/42/)

searching for KEYS948 ...

failed to match category for 'Konektory / Konektory pro přenos dat / Konektory RJ', select category:
  Electronics / Connectors
  Electronics / Connectors / Jumpers
  Electronics / Connectors / Audio Connectors
  Electronics / Connectors / Barrel Connectors
  Electronics / Connectors / Circular Connectors
> Enter Manually ...
  Skip ...
category name: RJ Connectors

failed to match some parameters from 'https://www.tme.eu/cz/details/keys948/konektory-rj/keystone/948/'

failed to match value for parameter 'Ethernet Category', select value:
  na PCB     | Mechanická montáž
  zásuvka    | Konektor
  úhlové 90° | Prostorová orientace
  8p8c       | Uspořádání vývodů konektoru
  RJ45       | Typ konektoru
  Enter Value Manually ...
> Skip ...
warning: failed to match 1 parameter from supplier API ('Ethernet Category')
successfully added TME part KEYS948 (http://inventree.rpi.lab.lan/supplier-part/43/)

the following parts are incomplete:
KEYS948
```
It asks for the same category mapping twice.

My `categories.yaml` now looks like this:
```yaml
            RJ Connectors:
                _description: RJ Modular Connectors
                _parameters:
                    - Connector Type
                    - Ethernet Category
                    - Number of Contacts
                _aliases:
                    - Konektory RJ
                    - Konektory RJ
```
Note the duplicated alias.

Version: `78fbee8`

## Expected behavior
When a category mapping is created, it should be immediately used for other parts in the same import command.

30350n, 2024-02-27:
Should be fixed!

ondras12345, 2024-02-27:
Thanks again for the quick fix!

closed 2024-02-27 by ondras12345