30350n/inventree-part-import #17: Value selector in "failed to match value for parameter" does not list all available values

ondras12345, 2024-02-27:
```console
$ inventree-part-import -o tme 54601-908WPLF
[...]
failed to match some parameters from 'https://www.tme.eu/cz/details/54601-908wplf/konektory-rj/amphenol-communications-solutions/'

failed to match value for parameter 'Mounting Type', select value:
> Amphenol Communications Solutions | Výrobce
  RJ45                              | Typ konektoru
  8                                 | Počet pinů
  zásuvka                           | Konektor
  Cat 3                             | Kategorie Ethernet
  Enter Value Manually ...
  Skip ...
Aborting Execution! (KeyboardInterrupt)
```
There is a parameter called "Elektrická montáž", but it is not listed by the value selector and thus an alias cannot be created interactively. After modifying my `parameters.yaml`:
```yaml
Mounting Type:
    _aliases:
        # ...
        - Elektrická montáž
```
and re-running the import, the parameter was matched correctly (no user interaction required).

30350n, 2024-02-27:
The interactive selection menus only show a limited number of options, sorted by closest matching (via fuzzy string matching). If you use multiple languages this obviously won't work well. Always showing all parameters/categories is not an option, as that'd clutter up the console too much. This could be a settable config variable though.

Any particular reason you don't just set your TME language to `en`? (Or alternatively: Your InvenTree category/parameter names to czech?)

ondras12345, 2024-02-27:
> Any particular reason you don't just set your TME language to `en`

Coincidentally, I have just done that. I did not do it initially because I needed TME links to be in Czech (otherwise, I have to go through the process of setting the language and currency every time I open one of them). I have written a custom hook that translates URLs, so now I am using `language: en`.

I do remember this happening in English, too, but it should be way less common.
Perhaps an "Enter parameter name manually" option could be added? That could make it possible to create the alias interactively. No need to list available parameters as it already prints a link to TME.

30350n, 2024-02-27:
Yeah, sounds good. Seemed easy to implement, so here we go. Would be great if you could test if it actually works 😅

30350n, 2024-02-27:
> I did not do it initially because I needed TME links to be in Czech (otherwise, I have to go through the process of setting the language and currency every time I open one of them). I have written a custom hook that translates URLs, so now I am using language: en.

Ah yeah, I get that, that's annoying indeed. The hook is also the perfect solution btw ^^

ondras12345, 2024-02-27:
It seems to be working perfectly. Thanks!

> The hook is also the perfect solution btw

It wasn't until I read your configs that I realized I could use a hook to modify the URL. It seems like best solution to me, too.

closed 2024-02-27 by ondras12345

30350n, 2024-02-27:
Yep, I really like the hooks :P
They just enable you to do anything you want on the input data!