30350n/inventree-part-import #19: Supplier choices are invalid when using external config dir

QuadratClown, 2024-03-05:
When using supplier constrained input flags like `-s` or `-o` together with the `-c` flag, the cli will not use the provided config dir when deciding which suppliers are valid choices.

# How to reproduce

- do not set up the default config dir
- set up the config in a local dir like `inventree-part-import`
  - set up a supplier, e.g. 'mouser'

From config folder, run:

`inventree-part-import -c $pwd -o mouser <arbitraryPart>` 

## Expected behaviour

Use config dir provided with `-c` for supplier choices

## Real behaviour

```
Error: Invalid value for '-o' / '--only': 'mouser' is not one of .
```

30350n, 2024-03-05:
Easiest way to fix this would be to simply also configure those suppliers in the default configuration. Alternatively you could also just **use** the default configuration directory, I mainly implemented the `-c` option to be able to quickly test things with different configurations.

QuadratClown, 2024-03-05:
Thanks for the quick answer. An alternative way could be to make the default location configurable. Not sure how hard that is with `click`.

The whole thing is by no means a show stopper anyway, it's just not quite obvious where the error is coming from when not yet familiar with the inner workings of the cli.

30350n, 2024-03-05:
No problem! Yeah the cause of the error is somewhat opaque.
Any particular reason why you aren't just using the default directory btw?

QuadratClown, 2024-03-05:
I'm using multiple devices with inventree and want to sync the config across them. On windows for example, the default folder is in `AppData/Local`, which I really dislike using for a purpose like this. Usually, I have just have a scripts folder that contain one folder per script with all necessary configs that I can just sync directly, platform independently.

30350n, 2024-03-05:
Ah okay, well I'm just using whatever `platformdirs.user_config_dir` spits out. You could also try to symlink your custom folder into the default location. Pretty sure that should work too.

QuadratClown, 2024-03-05:
The symlink approach is what I'm using for Kicad, guess then for here as well :) Thanks for your time!