30350n/inventree-part-import #24: Categories mapping ignores duplicate names from different paths

sle118, 2024-03-06:
Don't mean to bombard you with corner cases, but I did some analysis on an issue I was having with the parts category schema that matches Digikey. There are cases where sub categories are assigned different paths. A good example of this is for Transistors, which exists under both `Discrete Semiconductor Products/Transistors` and `Electronics/Transistors` which contains some sub categories as well. When building the category map, it seems like the 2nd instance of `Transistors` is ignored, which causes some import errors down the road. 

The issue happens here. 
```
    for category in categories.values():
        if category.structural or category.ignore:
            continue
        for alias in category.aliases:
            category_map[alias] = category
        if category.name not in ignore:
            if category.name not in category_map:
                category_map[category.name] = category
            else:
                ignore.add(category.name)
                category_map.pop(category.name)
```

Given that the map is keyed by name rather than by path, this is not a simple fix. So in the meantime, I will rename my duplicate categories in InvenTree and add aliases if I ever import parts that match one of the renamed categories.

thank you!

sle118, 2024-03-06:
> 
> Importing the complete DigiKey category structure into InvenTree is just an awful idea.

lol. the idea of a newbe for using the InvenTree system. I am starting to realize this now but I have a good amount of data entered and a goal in mind so cleanup will wait. In the meantime I renamed the duplicates to get around the issue and will select them as aliases when importing parts in that given category. At least this will document the issue if someone was to stumble on the same problem

closed 2024-03-07 by sle118

30350n, 2024-03-07:
(Will keep this open, as it's still a bug, but I don't really have time to tackle this, thus the wontfix tag).