30350n/inventree-part-import #28: TME GetPricesAndStocks rate limit is too conservative

ondras12345, 2024-03-10:
https://github.com/30350n/inventree_part_import/commit/f8ea691d72f4775f961fb3cfaf4e3764a39a013b introduced a limit to 1 request every 2.5 seconds. However, the e-mail TME sent says:
> We're reducing the maximum API requests to **two per second** for actions:
> - /Products/GetStocks
> - /Products/GetDeliveryTime
> - /Products/GetPrices
> - /Products/GetPricesAndStocks

Thus, I believe a limit of just over 0.5 s should be sufficient.

30350n, 2024-03-10:
Duh! I completely missread that in the E-Mail they sent and literally missread it again just now. That really took a while to register ...

Will reduce the limit to 0.6s. Did it actually affect you though? I would've assumed everything else would probably take >2s anyways.

ondras12345, 2024-03-10:
I did not actually test this, I noticed in git diff and thought you probably misread the e-mail.
Thanks for fixing it.

closed 2024-03-10 by ondras12345

30350n, 2024-03-10:
Gotcha, yeah I totally did 😅