30350n/inventree-part-import #31: Do not warn about part parameter templates from ignored categories and subcategories

ondras12345, 2024-03-17:
In `categories.yaml`, I have this:
```yaml
CNC:
  _ignore: True
```
I have manually defined several subcategories of CNC in InvenTree and added parameters to them. `inventree-part-import` now complains whenever it runs:
```console
$ inventree-part-import -o tme some_part 
(only 2 of 5 available supplier modules are configured)
successfully loaded 1 pre creation hooks!
warning: parameter template 'Shank Diameter' for 'CNC/Tools' on host is not defined in categories.yaml
warning: parameter template 'Number of Flutes' for 'CNC/Tools/End Mills' on host is not defined in categories.yaml
warning: parameter template 'Overall Length' for 'CNC/Tools' on host is not defined in categories.yaml
warning: parameter template 'Collet Size' for 'CNC/Collets' on host is not defined in categories.yaml
warning: parameter template 'Collet Type' for 'CNC/Collets' on host is not defined in categories.yaml
warning: parameter template 'Length of Cut' for 'CNC/Tools/End Mills' on host is not defined in categories.yaml
warning: parameter template 'Cutter Material' for 'CNC/Tools/End Mills' on host is not defined in categories.yaml
warning: parameter template 'Tip Diameter' for 'CNC/Tools' on host is not defined in categories.yaml
successfully setup categories!

searching for some_part ...
[...]
```
Is it possible to prevent this?

30350n, 2024-03-28:
Sorry for the late answer. This wasn't *super* straightforward (the code could probably use some more refactoring) but it should be fixed now!

ondras12345, 2024-03-28:
Thanks, there are no more warnings.

closed 2024-03-28 by ondras12345