30350n/inventree-part-import #32: import to part as supplier part

CarlosDerSeher, 2024-03-28:
Hey there great tool and really helpfull!

Maybe I am missing something but seems the tool can't do it. Here's what I try to achieve:

I got a part **C_100nF_0603_50V_5%_X7R** now it would be great to use inventree_part_import to actually get some Part Suppliers in there by using a command like this

`inventree-part-import -p C_100nF_0603_50V_5%_X7R 581-06035C104JAT4A `

where `-p` is the exisiting part where supplier data should be add.

Currently this doesn't seem possible and I have to either enter all the info manually or just don't use generic parts like C_100nF_0603_50V_5%_X7R

Do you think this is possible? Where do I need to look at to add this functionallity if you are not interested in doing it or just don't have time to do it?

30350n, 2024-03-28:
This is a similar issue/feature as #20. [My proposed solution](https://github.com/30350n/inventree_part_import/pull/20#issuecomment-1979741732) for that (adding `--merge-parts` and `--merge-manufacturer-parts` options) would also solve this, but I haven't found time to look into implementing it yet sadly.

You can also manually do this for now by importing the part, then linking the created ManufacturerPart to your existing Part and deleting the created Part you don't need/want.

CarlosDerSeher, 2024-03-28:
Alright thank you. I guess this can be closed then.

closed 2024-03-28 by CarlosDerSeher

CarlosDerSeher, 2024-04-02:
@30350n sorry to ask this here but I can't find this information in the documentation anywhere. How do you link ManufacturerPart and supplier part through the UI? I managed to do it using the admin interface by changing the "base part" in manufacturer and supplier part and then deleting the created part. Do you know if there a better/faster way to do this?

30350n, 2024-04-02:
That's exactly how you do it. It work t super straight forward, thus the `--merge-parts` functionality would definitely make things a little easier.

CarlosDerSeher, 2024-04-02:
Ok thank you.