30350n/inventree-part-import #33: AssertionError when multiple supplier parts exist with the same SKU

ondras12345, 2024-03-29:
1. Import a part
2. Manually create a new supplier part for it in inventree. Link it to a different supplier (company), but use the same SKU as the automatically imported supplier part.
3. Try to update the part by importing it again
4. Observe assertion error from this function:
  https://github.com/30350n/inventree_part_import/blob/14b62e821fc2bae64cdad8ed1589f4b07132bc0a/inventree_part_import/inventree_helpers.py#L23-L29

It seems to only match parts based on the SKU. I believe it should also use the supplier (company) - the caller function `import_supplier_part` does have that information available.
Btw the (incorrect) type annotation `inventree_api: InventreeCompany` suggests that this was perhaps planned at some point, but never got implemented.

30350n, 2024-03-31:
Should be fixed!

(I implemented this so that the company check only happens, if multiple SupplierParts with the same SKU exist. I think I prefer that behavior.)

ondras12345, 2024-03-31:
Thanks! Now it's working.

closed 2024-03-31 by ondras12345