30350n/inventree-part-import #40: Feature request: Parameter list in interactive is only five lines; might it be increased?

cmidgley, 2024-04-04:
## Summary

A proposal to increase the minimum number of properties shown in interactive mode when choosing a matching property and value.

## Background

When in interactive mode and importing a part that doesn't match the required parameters, a list of properties is shown that the user can select from, such as:

```
failed to match value for parameter 'FLASH Size', select parameter:
> 8MB PSRAM       | Memory Size
  TxRx + MCU      | Type
  56-QFN (7x7)    | Supplier Device Package
  Bluetooth, WiFi | RF Family/Standard
  -98.4dBm        | Sensitivity
  Match Parameter Manually ...
  Enter Value Manually ...
  Skip ...
  ```
  
Having the list with values is great, unless it's not in the first five lines.  Even with the fuzz match, I find it often misses the properties I need.  This results in a challenging workflow as you need to go to the vendor site, try to find the available vendor properties, and hand enter either the value (or better, the parameter name).  As a work-around, in my hooks.py file, I've added a `pprint.pprint(api_part)` as this at least dumps all values, but it's rather ugly!

## Solutions

The easiest solution is to increase the number of lines, such as in part_importer.py, function select_parameter():

```
N_MATCHES = min(20, len(parameters))
```

A better solution might be to have it in `config.yaml`.  I see there is a `max_results`, and it's only supported by `supplier_reichelt.py`, and since I don't use them I'm unsure what it does, but guessing using it for this purposes would be overloading it (so perhaps create a new `max_interactive_properties` value?)

I'm glad to make a pull, but thought I'd get your feedback on this prior to doing the work.

30350n, 2024-04-04:
I've probably glanced over that hardcoded '5' like a houndred times and thought to myself: I shoud really make this into a config parameters but never got around to it somehow. There's two new parameters (`interactive_category_matches` and `interactive_parameter_matches`) now, which allow you to set this in your config.

30350n, 2024-04-04:
Regarding `max_results`: I renamed this to `interactive_part_matches` as this reflects what it does better, it hasn't only been used by Reichelt (see here): https://github.com/30350n/inventree_part_import/blob/21810dc3c272042da7219e2feb8553b7a7efdb8b/inventree_part_import/part_importer.py#L74

But for Reichelt it's kind of relevant because I have to scrape once per search result (so you really want a small limit here).

cmidgley, 2024-04-06:
Tested and confirmed working.  Thanks!

closed 2024-04-06 by 30350n