30350n/inventree-part-import #41: Feature Request: Allow the part column name to be specified on cli/config to avoid prompting during multi-column file import

cmidgley, 2024-04-05:
## Background and use case

I am working on automating the workflow from my electronics design software to Inventree, including importing new parts.  I currently export a single multi-column file which is used for part creation with `inventree_part_import` and for BOM import into Inventree.  When using `inventree_part_import -i false somefile.csv` the user is required to manually select which column to use for the import, halting the automated process.  

This feature request is to be able to specify the column name in advance of import to avoid the prompt.

## User experience options

1) Add a CLI option that specifies the column name, such as `--import-column my-column-name`.  
2) Add a `config.yaml` option that specifies the column name.  
  _This may be too rigid, as importing a file using a different column name would require editing the config file._
3) A combination, where the config file specifies the default, but the CLI can override it including some syntax such as `--import-column false` or `--import-column default` to ignore it. 
  _While the most flexible approach, it feels like overkill for a low-use feature._
  
My recommendation is option 1.  I'm willing to do a pull-request if approved.

30350n, 2024-04-05:
Optional config parameters are super simple to add, that's why I generally prefer them to extra CLI options ^^

There's a new `auto_detect_columns` configuration option now (standard yaml list).

cmidgley, 2024-04-06:
Wonderful.  Confirmed working with both the new default with `part_id` as well as the ability to override it with `config.yaml`.

30350n, 2024-04-06:
Perfect, thanks for confirming ^^

closed 2024-04-06 by 30350n