30350n/inventree-part-import #42: Fix LCSC datasheets

zenermerps, 2024-04-13:
LCSC recently switched their datasheet links on the main product page to point to a PDF view with an ordering panel to its side. So the datasheet that now gets downloaded by the tool is actually html with .pdf extension. Please update the crawler to actually download the embedded pdf from the site instead of just the HTML.

30350n, 2024-04-14:
Thanks for reporting and also for presenting the fix right away! ^^

> For product page `https://www.lcsc.com/product-detail/Monitors-Reset-Circuits_LOWPOWER-LP5300B6F_C387703.html`, with "datasheet" link **`https://datasheet.lcsc.com/lcsc/1912111437_LOWPOWER-LP5300B6F_C387703.pdf`**

Hm I do actually seem to be getting a download redirect from that link too, but I guess the download function doesn't properly pick that up.

I'm replacing `"//datasheet.lcsc.com/"` with `"//wmsc.lcsc.com/wmsc/upload/file/pdf/v2/"` in LCSC datasheet urls now, could you quickly confirm that this works as intended and fixes the issue?

zenermerps, 2024-04-14:
Tried it with the example I gave above with current git master and it works now, thanks for the quick fix!

30350n, 2024-04-14:
Perfect, thanks for confirming!

closed 2024-04-14 by 30350n