30350n/inventree-part-import #44: Strip HTML tags from fields

zenermerps, 2024-04-14:
It seems LCSC has stray HTML tags in some fields, like in product listing https://www.lcsc.com/product-detail/LED-Display-Drivers_span-style-background-color-ff0-MBI-span-MBI5124GFN-B_C476807.html

This produces the following error:

```
❯ inventree-part-import MBI5124GFN-B
(only 3 of 5 available supplier modules are configured)
successfully loaded 17 pre creation hooks!
successfully setup categories!

searching for MBI5124GFN-B ...
(no results at DigiKey)
creating manufacturer '<span style="background-color:#ff0;">MBI</span>' ...
error: failed to import part with:
    name: ['Remove HTML tags from this value']

the following parts failed to import:
MBI5124GFN-B
```

30350n, 2024-04-15:
Ah, html tags in the manufacturer name, sure, that makes sense ...

(Should be fixed.)

zenermerps, 2024-04-15:
Works, thank you!

closed 2024-04-21 by 30350n