30350n/inventree-part-import #51: ModuleNotFoundError: No module named 'inventree_part_import.error_helper'

randrej, 2024-05-18:
You're vendoring in `error_helper` as a submodule, but it's not getting packaged and pushed to PyPI, hence this whole package doesn't run. Since you already packaged it on PyPI, just add it to the reqs.

For the AUR package I fixed it by just packaging and `find ... -exec sed`, but the fix should be here.

30350n, 2024-05-18:
Are you sure about that? It is present in the
[inventree_part_import-1.6.tar.gz](https://files.pythonhosted.org/packages/90/76/6005bb9fbfcd2b1ec2ffaaac3262bb8c313fef15714a9de70b7faa9c1d30/inventree_part_import-1.6.tar.gz) from https://pypi.org/project/inventree-part-import/#files.

The workflow I'm using is also configured to checkout submodules before building via hatch, so I don't see why it wouldn't be present in the wheel either: https://github.com/30350n/inventree_part_import/blob/22ff74053531fb4a7b4c8f5e01ede160f6927dc4/.github/workflows/publish.yaml#L12-L25

30350n, 2024-05-18:
Ah, TIL wheels are also just .zip files (could have expected that I guess ^^). And yeah `error_helper` is also present in the wheel.

30350n, 2024-05-18:
Seems like this is probably just a problem with your [PKGBUILD for the `-git` version](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-inventree-part-import-git). You should probably be using hatch there and also force checking out all submodules (not sure how that'll work exactly, never messed with Arch packages before).

randrej, 2024-05-18:
You seem to be correct, I'm pulling the source package (`https://files.pythonhosted.org/packages/source/i/inventree_part_import/inventree_part_import-1.6.tar.gz` as per the [Python packaging guidelines](https://wiki.archlinux.org/title/Python_package_guidelines)) and now I can't seem to replicate the issue, the folder is there; however, somehow the first time around the error_helper dir didn't end up in the package. Strange.

randrej, 2024-05-18:
> Seems like this is probably just a problem with your [PKGBUILD for the `-git` version](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=python-inventree-part-import-git). You should probably [...]

Yeah, this is true, there's a roundabout way to download submodules, but I didn't do it; I will, now.

randrej, 2024-05-18:
Yeah, false alarm; I can't seem to replicate it now. Will undo the find/sed thing.

closed 2024-05-18 by randrej

randrej, 2024-05-18:
I was looking at the wrong package, the -git one.