30350n/inventree-part-import #54: Add a way to skip parameters in categories

randrej, 2024-05-18:
Using your config, let's look at `Operational Amplifiers`. They're a subgroup of `Amplifiers`, which is a subgroup of `Integrated Circuits`. Since `Amplifiers` specify `Supply Current`, whenever you're importing an opamp (say, a `TL072`), you'll be asked for `Supply Current` which is not really something listed in opamp datasheets.

It'd be nice if there was an `_omit_parameters` field to ignore these things.

randrej, 2024-05-18:
Btw I'd like to help on things like this, I'm not just asking for you to do something, but you gotta approve it first.

30350n, 2024-05-19:
The categories and parameters in my config are mainly based on DigiKey's structure. And regarding that specific example, DigiKey indeed specifies `Supply Current` (without Load) for Opamps. For example for the [TL072HIDR](https://www.digikey.de/en/products/detail/texas-instruments/TL072HIDR/15205173). The value seems to come from the graphs on Page 28/29 of the [datasheet](https://www.ti.com/lit/ds/symlink/tl071h.pdf).

Anyways, regarding the feature suggestion: I have actually come across a case in my configuration where this could have been useful, but the format generally works without it. You'll just have to define the parameters further down the category tree.

I'm not per se against this feature, but I fear that it might be pretty confusing.

30350n, 2024-05-29:
After thinking about this a little more, I guess it wouldn't really hurt much to implement this and it might makes configuration easier in some cases.

Regarding the naming, `_omit_parameters` is already not bad. What do you think about something like `_omit_parent_parameters` or `_ignored_parent_parameters` (mainly to convey that this affects parameters inherited from parent categories)?

30350n, 2024-05-29:
(Oh also, offtopic: Did you receive my e-mail reply btw? I had some issues with the DNS settings for my mail server the first time I sent it ^^)

randrej, 2024-05-31:
Hey, I got your e-mail!

Yeah,  I agree with this, hopefully I'll find time soon to make an MR.

Also off topic here:

I'm also considering ways of incorporating the [template part / variant](https://docs.inventree.org/en/latest/part/template/#enable-template-part) functionality to have generic parts. I'll write an issue about it when I investigate it a bit further.

I'm currently doing an important project that uses your tool + Inventree, I want to see it through before suggesting any new functionality; I need to get a feel for how it works in real use.

30350n, 2024-06-01:
> Yeah, I agree with this, hopefully I'll find time soon to make an MR.

I might also be able to give it a shot if I find some time. I think this should require pretty minimal changes, like 10 lines or so at max.

> I'm also considering ways of incorporating the [template part / variant](https://docs.inventree.org/en/latest/part/template/#enable-template-part) functionality to have generic parts. I'll write an issue about it when I investigate it a bit further.

Definitely check out [#20 (comment)](https://github.com/30350n/inventree_part_import/pull/20#issuecomment-1979741732) if you do. That's the kind of functionality that I'd think makes the most sense here (probably with slightly different naming though).

> I'm currently doing an important project that uses your tool + Inventree, I want to see it through before suggesting any new functionality; I need to get a feel for how it works in real use.

Cool! Let me know if you run into any issues.

30350n, 2024-06-29:
This has been implemented in [`b28047d`](https://github.com/30350n/inventree_part_import/commit/b28047d2d526ee4b0b1615801dacfa5b1adcc6e7)!

closed 2024-06-30 by 30350n