30350n/inventree-part-import #56: LCSC not working

Zontex, 2024-06-25:
Trying as follows inventree-part-import C181133
gives me:

successfully setup categories!

searching for C181133 ...
warning: failed to get page, retrying in 3.0s with new session and user agent
(no results at LCSC)

reimporting failed/incomplete parts in interactive mode ...

searching for C181133 ...
(no results at LCSC)

the following parts failed to import:
C181133

Tried also with ID, names and other things.

30350n, 2024-06-25:
This was fixed in #48. If you [install the newest version from master](https://github.com/30350n/inventree_part_import/issues/49#issuecomment-2182638674) it will work.

closed 2024-06-25 by 30350n

Zontex, 2024-06-25:
Pip is outdated.
fixed it by : pip install git+https://github.com/30350n/inventree_part_import.git

30350n, 2024-06-25:
Yep, exactly. I haven't yet made a new release because mouser is still partially broken.