30350n/inventree-part-import #57: failed to connect to 'http://172.18.0.4:8000/' a local Docker InvenTree

MikeLemo1, 2024-08-18:
I'm just starting to play with and looking into InvenTree and Docker and I'm running the plugin config wizard interactively but am unable to connect to my docker hosted inventree instance which I'm able to connect to the browser with 
http://172.18.0.4:8000/
OR 
http://inventree.localhost/
But the wizard accept erroring:
 ```
error: failed to connect to 'http://172.18.0.4:8000/' with 'Could not connect to InvenTree server'
```

the ip was fetched with 
`docker ps -q | xargs -n 1 docker inspect --format '{{range .NetworkSettings.Networks}}{{.IPAddress}}{{end}} {{ .Name }}' | sed 's/ \// /'`

if it's relevant.

What am I doing wrong

30350n, 2024-08-22:
Hm not sure tbh, the error comes directly from the [`inventree-python`](https://github.com/inventree/inventree-python) module, so I could only imagine that it's some kind of version mismatch between your InvenTree version and it's version.

MikeLemo1, 2024-08-25:
I actually managed to connect to that address after all maybe the last digit was 5 but the interface behaved a bit differently than in [http://inventree.localhost/](http://inventree.localhost/) where the ip variant was a little buggier
And I still didn't figure out what is the address to connect to the postsql database from a db manager i.e. nvim-dbee.

30350n, 2024-08-25:
> I actually managed to connect to that address after all maybe the last digit was 5 but the interface behaved a bit differently than in http://inventree.localhost/ where the ip variant was a little [...]

Are you talking about the InvenTree web interface or `inventree-part-import` here?

MikeLemo1, 2024-08-25:
The web interface with the ip numbers vs of the wildcard addres but I want to connect to the main database of postgresql  inventree with an external db manager to search and adjust parameters. so I want to know what http url scheme to use and how to construct it. 
what he has there https://www.youtube.com/watch?v=MDlYsGbKJyQ&t=166s in 2:45

closed 2025-02-05 by 30350n