30350n/inventree-part-import #58: failed to import with the error 'API endpoint not found'

feijj, 2024-09-16:
inventree-part-import version 1.7

got the following error when running the command 

```shell
$ pipx run inventree-part-import RT9078-33GQZ -i true
warning: supplier 'mouser' (support level SCRAPING) is used ahead of supplier 'tme' (support level OFFICIAL_API)
(you might want to reorder them in 'suppliers.yaml')
(only 4 of 5 available supplier modules are configured)
successfully setup categories!

searching for RT9078-33GQZ ...
importing DigiKey part RT9078-33GQZTR-ND ...
error: failed to import part with: 
    detail: API endpoint not found

    url: http://localhost/api/part/category/None/

the following parts failed to import:
RT9078-33GQZ
``` 

I would appreciate any assistant on this issue. Thanks in advance!

30350n, 2024-09-16:
Hm, hard to tell what's the issue here. Try running with `--verbose`. You can also try reinstalling `inventree_part_import` (if you installed it a while ago).

atanisoft, 2024-10-15:
@30350n I suspect this is an Inventree version mismatch on the APIs. Here is a sample (also showing a Mouser failure):
```
inventree-part-import CAY16-1002F4LF --verbose
setting up InvenTree API ...
loading api configuration from 'inventree.yaml' ...
warning: supplier 'mouser' (support level SCRAPING) is used ahead of supplier 'lcsc' (support level INOFFICIAL_API)
(you might want to reorder them in 'suppliers.yaml')
(only 2 of 5 available supplier modules are configured)
setting up categories ...
successfully setup categories!

searching for CAY16-1002F4LF ...
searching at Mouser ...
(no results at Mouser)
searching at LCSC ...
importing LCSC part C840645 ...
error: failed to import part with: 
    detail: API endpoint not found

    url: http://localhost:1337/api/part/attachment/?part=466

FULL TRACEBACK:
Traceback (most recent call last):
  File "inventree-part-import/lib/python3.11/site-packages/inventree_part_import/part_importer.py", line 86, in import_part
    import_result |= self.import_supplier_part(supplier, api_part, existing_part)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/inventree-part-import/lib/python3.11/site-packages/inventree_part_import/part_importer.py", line 185, in import_supplier_part
    attachment_types = {attachment.comment for attachment in part.getAttachments()}
                                                             ^^^^^^^^^^^^^^^^^^^^^
  File "inventree-part-import/lib/python3.11/site-packages/inventree/base.py", line 471, in getAttachments
    return AttachmentSubClass.list(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "inventree-part-import/lib/python3.11/site-packages/inventree/base.py", line 203, in list
    response = api.get(url=url, params=kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "inventree-part-import/lib/python3.11/site-packages/inventree/api.py", line 530, in get
    response = self.request(url, method='get', **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "inventree-part-import/lib/python3.11/site-packages/inventree_part_import/retries.py", line 68, in request
    return super().request(api_url, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "inventree-part-import/lib/python3.11/site-packages/inventree/api.py", line 378, in request
    raise requests.exceptions.HTTPError(detail)
requests.exceptions.HTTPError: {'detail': 'Error occurred during API request', 'url': 'http://localhost:1337/api/part/attachment/', 'method': 'GET', 'status_code': 404, 'body': '{"detail": "API endpoint not found", "url": "http://localhost:1337/api/part/attachment/?part=466"}', 'headers': {'AUTHORIZATION': 'Token inv-......-20241015'}, 'params': {'part': 466}}

the following parts failed to import:
CAY16-1002F4LF
```

atanisoft, 2024-10-15:
I fixed this by force upgrading inventree pip package to the latest.