30350n/inventree-part-import #59: Initial Setup Broken

TimGoll, 2024-12-04:
I'm using the current release in pipx, which I think is `1.7`, I'm not sure how I get the exact version.

The problem is that the initial setup only generates the following folders/files:
[image]

In the same folders - I think - `inventree.yaml`, `config.yaml` and `suppliers.yaml` should be created. But this doesn't happen. When starting the tool for the first time, this is the output:
[image]

My guess is that the issue lies here:
https://github.com/30350n/inventree_part_import/blob/c044691bc03b3761d026e236f7dad252a531104a/inventree_part_import/config/__init__.py#L297-L298

TimGoll, 2024-12-04:
> > I'm using the current release in pipx, which I think is 1.7, I'm not sure how I get the exact version.
> 
> `inventree-part-import --version` should do the trick.

[image]

I don't think that parameter exists

TimGoll, 2024-12-04:
> This is expected behavior, config initialization happens when you first import a part. Calling `inventree-part-import` without any arguments won't trigger it.

Oh, then I misunderstood the readme. I thought the initial setup takes place before you import your first part.

30350n, 2024-12-04:
You are running a version <= 1.5 in that case. Try upgrading via `pipx upgrade inventree-part-import`.

30350n, 2024-12-04:
> Oh, then I misunderstood the readme. I thought the initial setup takes place before you import your first part.

Not really. It does take place before importing your first part, but you have to "try" to import a part to trigger it ^^

TimGoll, 2024-12-04:
> pipx upgrade inventree-part-import

But I just installed it today? Nevertheless, I tried it, this is the output:

[image]

30350n, 2024-12-04:
That's odd. And you only installed it via pipx?
Can you confirm that it's not available anymore if you `pipx uninstall` it? (And that it still doesn't default to 1.7 if you reinstall it again afterwards?)

TimGoll, 2024-12-04:
> That's odd. And you only installed it via pipx? Can you confirm that it's not available anymore if you `pipx uninstall` it? (And that it still doesn't default to 1.7 if you reinstall it again afterwards?)

I tried that before reporting that issue, but I will try it again

30350n, 2024-12-04:
Yeah then you might have installed it via normal pip or something? Check what `which inventree-part-import` outputs.

TimGoll, 2024-12-04:
> Yeah then you might have installed it via normal pip or something? Check what `which inventree-part-import` outputs.

[image]

Strange. but I tried it already in February of this year, but I'm absolutely sure that I used pipx at that point as well, this is why I used pipx uninstall today before reporting the issue. I have no idea where the pip package came from. But it seems to be resolved

TimGoll, 2024-12-04:
Thanks alot for your help! Really appreciated

closed 2024-12-04 by TimGoll

30350n, 2024-12-04:
No problem ^^

TimGoll, 2024-12-04:
Okay, I'm stuck again. I tried to access my inventree server via my local machine over this tool and failed. I assumed it was because of our company's proxies.

[image]

However, I tried the same thing with SSH access on the server as well and it failed again. Although with a different error this time:

[image]

I'm pretty sure I was able to log in February without any issues with the host+token. I'm confused why I need the user and password anyway. Also the documentation seems to suggest otherwise as well. I'm a bit lost.

30350n, 2024-12-04:
From the [README](https://github.com/30350n/inventree_part_import?tab=readme-ov-file#inventreeyaml):
> token: the user token to authenticate with (this will be retrieved automatically by the CLI)

You have to authenticate with your password once, then the token gets retrived from the server and stored in the `inventree.yaml`.

So if you think the problem lies with password auth, you can try providing the `host` and `token` in the `inventree.yaml` directly.

TimGoll, 2024-12-04:
Ahh, okay, I didn't get that. Personally I'd have preferred it to just enter the auth code. Thanks again!

TimGoll, 2024-12-05:
I am unable to make it work, I don't know why. I entered the URL and token manually into the file. This way it now actually tries to connect, but the authentication fails. Not sure why though, because it is the same token I use in KiCad.

[image]

Any ideas?

TimGoll, 2024-12-05:
Maybe as a feedback: The issue lies in inventree. We had two problems. The aforementioned Certs, but also the wrong url (I used the kicad api endpoint). Inventree did not check the output and silently failed at this point: 
https://github.com/inventree/inventree-python/blob/80ff58de0757cc6ce37a83bab0a1355819745746/inventree/api.py#L212

This happens because the return text is no json, but the login page HTML.

Maybe your tool should use the inventree logger and print it as well when enabling `verbose`. This would have helped us when debugging.

30350n, 2024-12-05:
> Maybe your tool should use the inventree logger and print it as well when enabling `verbose`. This would have helped us when debugging.

Good point! I was actually briefly thinking about that yesterday and wondering why it didn't do that already. It's implemented in [`bc09996`](https://github.com/30350n/inventree_part_import/commit/bc09996e96f7c9e5fbc50dd8c3ff50177f60001d) now.

TimGoll, 2024-12-06:
awesome, keep up with that good work!