30350n/inventree-part-import #60: hangs on uploading datasheet ...

cNavsky, 2024-12-30:
hangs on "uploading datasheet ...". no timeout, only way to break is to terminate session.
Happens on some parts, like this (tried multiple times):
https://www.digikey.ca/en/products/detail/molex/0039012040/61376

same for Eaton parts:
BK/S-8202-2-R
BK/S-8202-1-R
BK/ABC-20-R

30350n, 2025-01-02:
Thanks for reporting!

Can you try running the version from `master` with `--verbose`? That should log any InvenTree API calls to the console and might help identifying where the issue lies exactly.

cNavsky, 2025-01-02:
Hello,

year, tried that, no additional output at console it runs from. 
Hangs on this part as well: https://www.digikey.ca/en/products/detail/0472192001/3044147

Here is what I see on console:

~ % inventree-part-import --verbose 0472192001
setting up InvenTree API ...
loading api configuration from 'inventree.yaml' ...
(only 2 of 5 available supplier modules are configured)
setting up categories ...
successfully setup categories!

searching for 0472192001 ...
searching at DigiKey ...
importing DigiKey part WM6698TR-ND ...
uploading datasheet ...

and here is the inventree-server output log from docker:

2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/ HTTP/1.1" 200 1338 "-" "python-requests/2.32.3"
2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/user/me/ HTTP/1.1" 200 153 "-" "python-requests/2.32.3"
2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/company/23/ HTTP/1.1" 200 366 "-" "python-requests/2.32.3"
2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/company/4/ HTTP/1.1" 200 369 "-" "python-requests/2.32.3"
2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/part/category/ HTTP/1.1" 200 42127 "-" "python-requests/2.32.3"
2025-01-02 12:09:48 172.18.0.1 - - [02/Jan/2025:20:09:48 +0000] "GET /api/part/parameter/template/ HTTP/1.1" 200 11578 "-" "python-requests/2.32.3"
2025-01-02 12:09:49 172.18.0.1 - - [02/Jan/2025:20:09:49 +0000] "GET /api/part/category/parameters/ HTTP/1.1" 200 171374 "-" "python-requests/2.32.3"
2025-01-02 12:09:49 172.18.0.1 - - [02/Jan/2025:20:09:49 +0000] "GET /api/part/parameter/template/ HTTP/1.1" 200 11578 "-" "python-requests/2.32.3"
2025-01-02 12:09:50 172.18.0.1 - - [02/Jan/2025:20:09:50 +0000] "GET /api/company/part/?SKU=WM6698TR-ND HTTP/1.1" 200 2 "-" "python-requests/2.32.3"
2025-01-02 12:09:50 172.18.0.1 - - [02/Jan/2025:20:09:50 +0000] "GET /api/company/part/manufacturer/?MPN=0472192001 HTTP/1.1" 200 158 "-" "python-requests/2.32.3"
2025-01-02 12:09:50 172.18.0.1 - - [02/Jan/2025:20:09:50 +0000] "GET /api/part/366/ HTTP/1.1" 200 1216 "-" "python-requests/2.32.3"
2025-01-02 12:09:51 172.18.0.1 - - [02/Jan/2025:20:09:51 +0000] "GET /api/attachment/?model_type=part&model_id=366 HTTP/1.1" 200 2 "-" "python-requests/2.32.3"
2025-01-02 12:09:56 172.18.0.1 - - [02/Jan/2025:20:09:56 +0000] "GET /api/notifications/?read=false&limit=1 HTTP/1.1" 200 52 "http://10.0.1.58/platform/settings/admin/plugin" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15"
2025-01-02 12:10:26 172.18.0.1 - - [02/Jan/2025:20:10:26 +0000] "GET /api/notifications/?read=false&limit=1 HTTP/1.1" 200 52 "http://10.0.1.58/platform/settings/admin/plugin" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15"

after that it keeps repeating:
 "GET /api/notifications/?read=false&limit=1 HTTP/1.1" 200 52 "http://10.0.1.58/platform/settings/admin/plugin" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15"

hope this helps.

closed 2025-01-02 by cNavsky

palandri, 2025-02-19:
I get this same behaviour (block on uploading datasheet) on some parts, specifically from Digikey.

My workaround is to import the part from Mouser (using --only=mouser) and then, fetch prices from Digikey (omitting --only or using --only=digikey).

tlf30, 2025-11-04:
I have the same issue. Perhaps a timeout after like 10 seconds just skip the datasheet, with an argument to either skip the datasheet or change the timeout. 

```
DEBUG:urllib3.connectionpool:https://api.digikey.com:443 "GET /products/v4/search/P4.99KHDKR-ND/productdetails HTTP/1.1" 200 7157
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): inv.tlf.io:443
DEBUG:urllib3.connectionpool:https://inv.tlf.io:443 "GET /api/company/part/?SKU=P4.99KHDKR-ND HTTP/1.1" 200 2
importing DigiKey part P4.99KHDKR-ND ...
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): inv.tlf.io:443
DEBUG:urllib3.connectionpool:https://inv.tlf.io:443 "GET /api/company/part/manufacturer/?MPN=ERJ-3EKF4991V HTTP/1.1" 200 217
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): inv.tlf.io:443
DEBUG:urllib3.connectionpool:https://inv.tlf.io:443 "GET /api/part/6/ HTTP/1.1" 200 1282
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): inv.tlf.io:443
DEBUG:urllib3.connectionpool:https://inv.tlf.io:443 "GET /api/attachment/?model_type=part&model_id=6 HTTP/1.1" 200 2
uploading datasheet ...
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): industrial.panasonic.com:443
```

PS: Thank you so much for this too, it is a life saver.

Kjubyte, 2026-01-16:
I couldn't download Molex datasheets. After changing the User-Agent in [/inventree_part_import/inventree_helpers.py#L169](https://github.com/30350n/inventree-part-import/blob/master/inventree_part_import/inventree_helpers.py#L169) to `Mozilla/5.0 (X11; Linux x86_64; rv:146.0) Gecko/20100101 Firefox/146.0` it works now.

Maybe it would be a good idea to make the User-Agent configurable? I guess it's some sort of scraping protection. curl also doesn't work with it's default User-Agent.