30350n/inventree-part-import #61: `AttributeError` after chaning `categories.yaml`

schw4rzlicht, 2025-01-02:
I added a category to InvenTree and ran part import afterwards. Part import complained about the new category not being configured in `categories.yaml`, so I added it according to the schema (side note - maybe it'd be a good idea to import that category rather than complaining?) but noticed that part import crashes with this stack:

```
Traceback (most recent call last):sives/Test' ...
  File "/Users/xx/.local/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
             ~~~~~~~~~~~~~~~~~~~~~^^
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree_part_import/cli.py", line 23, in wrapper
    func(*args, **kwargs)
    ~~~~^^^^^^^^^^^^^^^^^
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree_part_import/cli.py", line 194, in inventree_part_import
    importer = PartImporter(inventree_api, interactive=interactive == "true", verbose=verbose)
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree_part_import/part_importer.py", line 41, in __init__
    self.category_map, self.parameter_map = setup_categories_and_parameters(self.api)
                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree_part_import/categories.py", line 45, in setup_categories_and_parameters
    part_category = PartCategory.create(inventree_api, {
        "name": category.name,
    ...<2 lines>...
        "parent": parent.pk if parent else None,
    })
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree/base.py", line 204, in create
    return cls(api, data=response)
  File "/Users/xx/.local/pipx/venvs/inventree-part-import/lib/python3.13/site-packages/inventree/base.py", line 82, in __init__
    pk = data.get(self.getPkField(), None)
         ^^^^^^^^
AttributeError: 'list' object has no attribute 'get'
```

That also happens when I just rename one of the existing categories without adding new ones.

macOS Sonoma 14.6.1
Python 3.13

schw4rzlicht, 2025-01-02:
Same thing happens when category cannot be matched, choosing "Enter Manually" and e.g. typing in "Electronics" - which exists in InvenTree and `categories.yaml` (which in this case is the default one from auto setup with default categories).

30350n, 2025-01-02:
Without looking into it deeply and just glancing at the error: This looks like the return type of an API call got changed from `dict` to `list`. (So it's showing up now because of your relatively new InvenTree version. I really have to update this to work on newer ones, but I won't be able to look into this further until February most likely.)

closed 2025-01-02 by schw4rzlicht

30350n, 2025-01-02:
That's kind of weird but good for me I guess. Glad you got that figured out.

If you create the `inventree-python` issue, feel free to mention it here.