30350n/inventree-part-import #62: mouser scraping blocked

galmiklos, 2025-02-03:
Hi,

I am trying to import Mouser parts, but none of the parameters are matched, I keep getting warnings, like this.

```
warning: failed to finalize part specifications from 'https://www.mouser.com/ProductDetail/YAGEO/RC0603FR-07150KL?qs=VU8sRB4EgwAs5t5QrX5iHA%3D%3D' (blocked)
warning: failed to match 6 parameters from supplier API ('Rated Power', 'Mounting Type', 'Tolerance', 'Resistance', 'Composition', 'Package Type')
```

Looking at the code, and the Mouser API, I think I understand how it is supposed to work.

Firstly, the Mouser API gives only a few "ProductAttributes", so apparently "scraping" is attempted to get all the parameters, at the "ProductDetailUrl". The problem is, that that URL does not seem to work when python is trying to connect.

In supplier_mouser.py:

```
        url = api_part.supplier_link
        if not (result := scrape(url, fallback_domains=FALLBACK_DOMAINS)):
            warning(f"failed to finalize part specifications from '{url}' (blocked)")
            return True
```

This is the error message curl throws.

```
gal@roki:~/.config/inventree_part_import$ curl https://www.mouser.com/ProductDetail/YAGEO/RC0603FR-07150KL?qs=VU8sRB4EgwAs5t5QrX5iHA%3D%3D
curl: (92) HTTP/2 stream 0 was not closed cleanly: INTERNAL_ERROR (err 2)
```

If I copy&paste the URL to a browser, I get to the correct product detail page.

Has anyone found a solution to this by chance? Anything I googled didn't help.

Thanks,
Miki.

galmiklos, 2025-02-04:
I added a few lines to scrape.py to debug http traffic, and this is what I saw.

```
(inventree-part-import) gal@roki:~/.config/inventree_part_import$ inventree-part-import --supplier mouser 603-RC0603FR-07150KL
(only 1 of 5 available supplier modules are configured)
successfully loaded 17 pre creation hooks!
successfully setup categories!

searching for 603-RC0603FR-07150KL ...
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): www.mouser.com:443
send: b'GET /ProductDetail/YAGEO/RC0603FR-07150KL?qs=VU8sRB4EgwAs5t5QrX5iHA%3D%3D HTTP/1.1\r\nHost: www.mouser.com\r\nUser-Agent: Mozilla/5.0 (iPhone; CPU iPhone OS 17_6_1 like Mac OS X) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.6 Mobile/15E148 Safari/604.1\r\nAccept-Encoding: gzip, deflate\r\nAccept: */*\r\nConnection: keep-alive\r\nAccept-Language: en-US,en\r\n\r\n'
reply: 'HTTP/1.1 403 Forbidden\r\n'
header: Accept-CH: Sec-CH-UA,Sec-CH-UA-Mobile,Sec-CH-UA-Platform,Sec-CH-UA-Arch,Sec-CH-UA-Full-Version-List,Sec-CH-UA-Model,Sec-CH-Device-Memory
header: Charset: utf-8
header: Content-Type: text/html;charset=utf-8
header: Access-Control-Allow-Credentials: true
header: Access-Control-Expose-Headers: x-dd-b, x-set-cookie
header: Strict-Transport-Security: max-age=31536000;
header: X-DataDome: protected
header: X-XSS-Protection: 1; mode=block
header: Access-Control-Allow-Origin: *
header: X-Content-Type-Options: nosniff
header: X-DataDome-CID: AHrlqAAAAAMAGrqqXkfBXCQAL7lOSA==
header: X-FRAME-OPTIONS: sameorigin
header: X-DD-B: 3
header: X-Akamai-Transformed: 0 - 0 -
header: Expires: Tue, 04 Feb 2025 05:01:11 GMT
header: Cache-Control: max-age=0, no-cache, no-store
header: Pragma: no-cache
header: Date: Tue, 04 Feb 2025 05:01:11 GMT
header: Connection: close
header: Set-Cookie: datadome=UXrpvrQw8Wr1vYPDV6aEY3lYxcI5kNuJ_06wh2oC1amUzMG~f9LAuBrqa7Kcq0jhhgo79DQBb9Y3xZazizkaKXXJvONvH1rpdp7H8pUW05iCWgUFil3YmwTVzoN6bYSo; Max-Age=31536000; Domain=.mouser.com; Path=/; Secure; SameSite=None
header: Server-Timing: cdn-cache; desc=MISS
header: Server-Timing: edge; dur=25
header: Server-Timing: origin; dur=25
header: PR_ORIGIN_ID: DC1-DC2-WLB
header: PR_Rule_Name: d2b74f152512397
header: PR_Rule_Number: 7
header: Set-Cookie: akacd_Default_PR=3916098070~rv=55~id=48bf14a73220674e2ef4c50f4107bf76; path=/; Secure; SameSite=None
header: Akamai-GRN: 0.253a2f17.1738645271.3a7a0a61
header: Set-Cookie: bm_ss=ab8e18ef4e; Secure; Domain=.mouser.com; Path=/; HttpOnly; Max-Age=3600
header: Set-Cookie: _abck=AA0EAC823DF83D6BB72C7F41A66E7943~-1~YAAQJTovF9LsH7uUAQAACnNVzw15oyDANHBLzvAohTQi8/WYa2UrxpzmcUMU5R1A+v4SOPLvCEL0ePlJEE6p9VxQ1uvwmqW2jW4RuhyDuxqZm6Jxo1tqSdCG1bBIxlimySMD1xVnOY0W1LcKLxAmX5yMC9fsN/bxllb3upSfPbMjhD6g9Qhrbj+HaryVICyKwKwyPNhkKlcx41hKaO3SpN+tRm+AMb+GHnzN9Ks7qNZWFHnkPpwUT/cmOl4JG0bU7t+P+dnZdVfGG12rceWcjdZ1k3fM3ft3h3Ek1y8TmhgQ+GiDix1wvAyVnAQHVmum4LZGgfdcqI+kjj7W+nfouTgq35WjnqJ2DyccbQkL3wkfeEePpue2fgjycpLrfPHsdHAeJpFMFhJIN4IapZU1qJq2Wm4O0DrGxlehWDcO~-1~-1~-1; Domain=.mouser.com; Path=/; Expires=Wed, 04 Feb 2026 05:01:11 GMT; Max-Age=31536000; SameSite=None; Secure
header: Set-Cookie: bm_mi=1168911CBA892EF72C0EFAFFE788F889~YAAQJTovF9PsH7uUAQAACnNVzxohkbd82adtPVG7KXMBhXklJeX5iM/LzXduQi4rvk+vRS27qCRU5QIqZrC9K07uUUweQZMh8u8j4CDDq6ndJb923cvtCLZ3nZVJJt2pRAQ4sGIOf1DPYiH7iXSVdr7/65VofSqcNRFfC22zeV5jll8/MeInh01Ro43K60rNKi6KMPcSR0wNCWJWt0JWugWC0XyyoUC08VBLXI48ccF96vn/6zEeztPenaZpK4RzP0Fr0q75j3tNVfiUTq+o/uMUPsZuZgTCop/NlWxq38Gq/F/Lklr0T+zBfoaadUtgu2mfLI0MuJEWaZCtzGv9JqjPBvRg9Tn21QLS8ooCdkVnxA==~1; Domain=.mouser.com; Path=/; Expires=Tue, 04 Feb 2025 05:01:11 GMT; Max-Age=0; SameSite=None; Secure
header: Set-Cookie: bm_s=YAAQJTovF9TsH7uUAQAACnNVzwLu8z2LZkh2AEijATRbFa58H9IPDXBuU2PTXomIbH6uHQLWQyN1xmLJD8VvBHih0TqCcTcUsSaFpfJkxYQyWeqtOFutwNZYbz+nKMx6Un64iKbkXgPKa7oNPDSGdKR2jwmPo7H/cunXwTPHxBjyqxRY17cq7MI+6Xs0jLAW/3jEfZU8n7OFQGc3qnS0P0ofkCKSQS6JXCUDiZzFwakGuw7yjMPdAZgE2u1uQ6mLt9pv//C7eFhGLXZPhD5vWHaGOeh5RmbYhmyoznaVLKLVKV5JAVaL4LEKJI4wPM4DBSubLjyPdUMSKD1b8Sc1QkwM0JLlrfqAEZqTGzULOU4kwYqiWF/kIwWI1jN5ZM687S54yOeS195guc5EAe33lqxdYLP7esKHhXAAt6Ec/+/bDKrpaKNOt4DNzpUr9laTXYeezqup8x6dEw==; Domain=.mouser.com; Path=/; Expires=Fri, 07 Mar 2025 05:01:11 GMT; Max-Age=2678400; SameSite=None; Secure; HttpOnly
header: Set-Cookie: bm_so=511D4E9ACC93FBA80AE90C866B1B77556BCADFDA4768B5C2393072409BDC0D9A~YAAQJTovF9XsH7uUAQAACnNVzwLOD9dCIbMTv4Nw3OST7XN6K+ryBYOyvGalLjtJ3bhe4L6LMwGdETzaH/l3qpRm5KZzTeAss+w4ckhoYHyYIy5Lece0BqQaMudV+/i0tC51VUjcc9OTaEWHKMbj0C+3pHHG6AIsHGOrYIPqCVhnBpNvh0Kt1QqP6YUsUc1tprZDK7ykQeTSQ36395GgqBz+cONtvXqJWDcEtRZTSw/7mfxauXzjNoBK8eWP5jpC9Q7hxCOJrqLpo0UP8CbyZfjn2kn8opiASj6onWIqMD8W8JmaKKrnZ8efmoRVAeq9PJxWNAYUjVRjAOqWIiVnBGlIkx6xNKDY47Wk5cSno1745s1iBX1RprpjRw/adS8R1KmZVx0DvbIjK/1UnJ5V0aScoVDtIaG5Qa431VOnmqnT9SWd5JASGj+YxtSbS5NW/Ef39omcATg272EciQ==; Domain=.mouser.com; Path=/; Expires=Wed, 05 Feb 2025 05:01:11 GMT; Max-Age=86400; SameSite=None; Secure
header: Set-Cookie: bm_sz=AF14B51263E407668392AA3B69B6205A~YAAQJTovF9bsH7uUAQAACnNVzxqkxGx4l7Z8SRlcoUYzN+dAdLJC9ml/GPuMIf6Xox+WsrP8xeTUVp0AJQXbA1SQ9Cj0oMP1YA7qWh2La45f7KPS4MizbDZLqeeSMQgTVr1b/w16TJlKclF/L17obG1ufevJY0th/BQezZgPKRirYTUr8QdKkxYLaNYowEAAu9sV1oB2DVGeRzrJqSqRx6KD0+WpeAU9a5LN2SOl65Dfx06KNEnoeIvF5yUL7e5ZCdXG1A9V/1MZQzVqYNq5zLFIDUC8q/mP2IGMO2lBpHGH5WSeVEou1k0CIMXJghGEnjhWOQgETi+JklKm6LOga3zq2y/J3/vg4UZdDg==~4277826~3225911; Domain=.mouser.com; Path=/; Expires=Tue, 04 Feb 2025 09:01:11 GMT; Max-Age=14400; SameSite=None; Secure
header: Server-Timing: ak_p; desc="1738645271190_388971045_981076577_4977_21789_2_24_-";dur=1
DEBUG:urllib3.connectionpool:https://www.mouser.com:443 "GET /ProductDetail/YAGEO/RC0603FR-07150KL?qs=VU8sRB4EgwAs5t5QrX5iHA%3D%3D HTTP/1.1" 403 None
warning: failed to finalize part specifications from 'https://www.mouser.com/ProductDetail/YAGEO/RC0603FR-07150KL?qs=VU8sRB4EgwAs5t5QrX5iHA%3D%3D' (blocked)
```

Other than "HTTP/1.1 403 Forbidden", I don't see anything revealing.

galmiklos, 2025-02-04:
After analyzing the request in Firefox, I managed to put together a curl command, that Mouser actually responded with an html payload, not just the HTTP 403 error.

```
(inventree-part-import) gal@roki:~/.config/inventree_part_import$ curl -v -H "User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:134.0) Gecko/20100101 Firefox/134.0" -H "Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8" -H "Accept-Language: en-US,en;q=0.5" -H "Accept-Encoding: gzip, deflate, br, zstd" -H "Connection: keep-alive" 'https://www.mouser.com/ProductDetail/Vishay-Dale/CRCW06031K50FKEA?qs=TPnjyyNsHsazZOiw0hmO4g%3D%3D'
*   Trying 96.16.17.77:443...
* Connected to www.mouser.com (96.16.17.77) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Finished (20):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.2 (OUT), TLS header, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use h2
* Server certificate:
*  subject: jurisdictionC=US; jurisdictionST=Delaware; businessCategory=Private Organization; serialNumber=4283470; C=US; ST=Texas; L=Mansfield; O=Mouser Electronics, Inc.; CN=www.mouser.com
*  start date: May 14 00:00:00 2024 GMT
*  expire date: Jun 13 23:59:59 2025 GMT
*  subjectAltName: host "www.mouser.com" matched cert's "www.mouser.com"
*  issuer: C=US; O=DigiCert Inc; OU=www.digicert.com; CN=GeoTrust EV ECC CA 2018
*  SSL certificate verify ok.
* Using HTTP2, server supports multiplexing
* Connection state changed (HTTP/2 confirmed)
* Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* Using Stream ID: 1 (easy handle 0x5b11bfea2eb0)
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> GET /ProductDetail/Vishay-Dale/CRCW06031K50FKEA?qs=TPnjyyNsHsazZOiw0hmO4g%3D%3D HTTP/2
> Host: www.mouser.com
> user-agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:134.0) Gecko/20100101 Firefox/134.0
> accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
> accept-language: en-US,en;q=0.5
> accept-encoding: gzip, deflate, br, zstd
> connection: keep-alive
> 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< HTTP/2 403 
< accept-ch: Sec-CH-UA,Sec-CH-UA-Mobile,Sec-CH-UA-Platform,Sec-CH-UA-Arch,Sec-CH-UA-Full-Version-List,Sec-CH-UA-Model,Sec-CH-Device-Memory
< charset: utf-8
< content-type: text/html;charset=utf-8
< access-control-allow-credentials: true
< access-control-expose-headers: x-dd-b, x-set-cookie
< strict-transport-security: max-age=31536000;
< x-datadome: protected
< x-xss-protection: 1; mode=block
[... 17 lines cut ...]
< pr_origin_id: DC1-DC2-WLB
< pr_rule_name: d2b74f152512397
< pr_rule_number: 7
< set-cookie: akacd_Default_PR=3916127517~rv=84~id=6b35152acd08fdfa762fabae25225375; path=/; Secure; SameSite=None
< akamai-grn: 0.253a2f17.1738674718.3e8e8c08
< set-cookie: bm_ss=ab8e18ef4e; Secure; Domain=.mouser.com; Path=/; HttpOnly; Max-Age=3600
< set-cookie: _abck=3FFB5D32977FA952297A5D8AAB9E73C9~-1~YAAQJTovF/yBPLuUAQAAjsYW0Q3NCSIHljFuH08P/4tB6draPDtCy0gwnEGlgR2YacdDTKF53DjV0x5cQ8lrIUYKuXI4hIOjtCDRvIUaxpFGT6/E/QCVbInJJATyAX4sFbmOBAgvE4GpyKhP89UHRfyMcxlkYcwE6KP6LIxPfHcg1w3wLAhFJ0ESlvzJJl8N520F2cI5yxShnapq6gf7LGhN+lskVzldBSXhJkSzt6TE2uYjCeDvscQXr3QFoynWW7k32xLaB231F0sB7XDp+uL8fIrvLrjevjRi7JBgiu7ySMhfWjyuC+8FKEFvKEPuJphIyo29aEILht8ZXF1lG5MlEjMKrm8j217Omi7LpcRyiPPBMPrULYj1NBKr2mijSbbvHkuDHOHvfWMoKz6dEYNqpqVW1Zhhouv+c0yf1BcKOPLJpOo=~-1~-1~-1; Domain=.mouser.com; Path=/; Expires=Wed, 04 Feb 2026 13:11:58 GMT; Max-Age=31536000; SameSite=None; Secure
< set-cookie: bm_mi=30ABFA7254F056BC4D87A16639D6AF0C~YAAQJTovF/2BPLuUAQAAjsYW0RoWAU/b09NNNzNxPC8+3DlMFK5GhBSQ8FVrFzOpg++uG9hKGu3lMw7QJ1AWfTVaS/AeLSG41zHL1PM7SPsZARhXa8bwAtAjoxhGCrGEGQiCz8CGxlu4O87NMH3yRR+NoTioITZ33XiuJcoqdvtEZ4TkNsXNo077WFQf2IA/UAvVmRlA3eKDOREQIGRz4t/EvL9m4XUxBH1K8QIepBr3H1XF9YygjYelihXBqpuKQYnt3LdfzHYQWiHcuU5Ll02u+AxbmbL67BYy0w8wqKm48XRbBBFEkyih/3KGV9OkloL3hj7rlA6TkRhPSkrH+Xc1u8/opVbuAtwVIOU6wrmMu9i6rJb+ZA==~1; Domain=.mouser.com; Path=/; Expires=Tue, 04 Feb 2025 13:11:58 GMT; Max-Age=0; SameSite=None; Secure
[... cut ...] mport)
```
I saved that html to a file, and opened the file in a browser, and this is what I got.

[image]

So, as far as I can see, Mouser does not allow access to that URL from a program, and it is a security measure from their part. So, I guess, other than trying to become a black heat hacker, using that URL to try to scrape the parameters missing from the API, will not work. :-(

I will try to ask Mouser if they would consider adding more information to the search API response. Wish me luck.

30350n, 2025-02-04:
> Firstly, the Mouser API gives only a few "ProductAttributes", so apparently "scraping" is attempted to get all the parameters, at the "ProductDetailUrl". The problem is, that that URL does not seem to work when python is trying to connect.

Exactly, the Mouser API doesn't offer any information about product parameters sadly, that's why scraping their website directly is required to retrieve them.

> I will try to ask Mouser if they would consider adding more information to the search API response. Wish me luck.

I did exactly this a couple of months ago and after having e-mail conversations with a bunch of people they denied all my requests.
It'd be great if you could still also give it a shot. Maybe they'll fix the API if enough people complain 🤷

galmiklos, 2025-02-05:
I found a simple workaround to this problem. As I start understanding the principles of InvenTree a little bit more (it's been only I few days I installed it, and started using it) I realized, you don't have to create the part from Mouser with all the parameters. You can search by manufacturers' part number with digikey or lcsc, that will create the part with all the parameters, and when inventree-part-import finally gets to searching Mouser, the part will already be there, and only a Mouser manufacturer's part will be added, For that, the Mouser search API gives enough information. You just have to make sure it is not Mouser you query first, which you can do with the --supplier option.

30350n, 2025-02-05:
I wouldn't exactly call that a workaround, as it still doesn't solve the problem for parts that are only available at Mouser, but not at other suppliers, but yes, that's how `inventree-part-import` generally works. Part parameters only get retrieved from the first supplier that returns a part.

As you already noticed, you can set supplier that gets queried first by using the `--supplier` option, alternatively you can also just edit the order of suppliers in the `suppliers.yaml` config file to change their default order. I'd recommend doing this.

I've also experimented with UserAgent headers and using browser cookies yesterday and managed to crawl Mouser relatively consistently by using `iOS` user agents in combination with cookies from my browser after loading the Mouser website there once (see the 2 commits above). We'll have to see if/how long that keeps working though.

30350n, 2025-02-05:
I've just released [1.8](https://github.com/30350n/inventree_part_import/releases/tag/1.8) which contains these changes.

galmiklos, 2025-02-11:
I added 'firefox' as browser_cookies in suppliers.yaml to both lcsc and mouser, and tried to import a random mouser part. It failed like this.

```
(inventree-part-import-1.8) gal@roki:~$ inventree-part-import -s mouser 941-CAB5R0A23GM4T
(only 3 of 5 available supplier modules are configured)
successfully loaded 17 pre creation hooks!
successfully setup categories!

searching for 941-CAB5R0A23GM4T ...
warning: failed to get page, retrying in 3.0s via 'www2.mouser.com' with new session and user agent
warning: failed to get page, retrying in 3.0s via 'eu.mouser.com' with new session and user agent
warning: failed to finalize part specifications from 'https://www.mouser.com/ProductDetail/Wolfspeed/CAB5R0A23GM4T?qs=iLKYxzqNS74tc9ZdnsyFwA%3D%3D' (blocked)
error: failed to match category for 'Discrete Semiconductor Modules'
(no results at DigiKey)
Traceback (most recent call last):
  File "/home/gal/inventree-part-import-1.8/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/cli.py", line 23, in wrapper
    func(*args, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/cli.py", line 210, in inventree_part_import
    importer.import_part(part, None, supplier, only_supplier)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 64, in import_part
    results, result_count = async_results.get()
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 774, in get
    raise self._value
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/suppliers/base.py", line 95, in cached_search
    return self.search(search_term)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/suppliers/supplier_lcsc.py", line 28, in search
    search_result = self.scrape(SEARCH_URL.format(search_term))
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/suppliers/base.py", line 117, in scrape
    self._setup_session()
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/suppliers/base.py", line 177, in _setup_session
    self.setup_hook()
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/inventree_part_import/suppliers/supplier_lcsc.py", line 145, in setup_hook
    self.session.get(CURRENCY_URL.format(self.currency), timeout=self.request_timeout)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/requests/sessions.py", line 602, in get
    return self.request("GET", url, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/requests/adapters.py", line 664, in send
    timeout = TimeoutSauce(connect=timeout, read=timeout)
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/urllib3/util/timeout.py", line 115, in __init__
    self._connect = self._validate_timeout(connect, "connect")
  File "/home/gal/inventree-part-import-1.8/lib/python3.10/site-packages/urllib3/util/timeout.py", line 159, in _validate_timeout
    raise ValueError(
ValueError: Attempted to set connect timeout to 0.0, but the timeout cannot be set to a value less than or equal to 0.
```
I have the request_timeout set in config.yaml.

```
gal@roki:~/.config/inventree_part_import$ cat config.yaml
currency: USD
language: en
location: US
scraping: true
datasheets: link
interactive: twice
interactive_part_matches: 10
request_timeout: 15.0
retry_timeout: 3.0
```

galmiklos, 2025-02-11:
If I change `timeout=self.request_tiimeout` to `timeout=10.0` in supplier_lcsc.py in line 145, it works. Apparently self.request_timeout is not initialized there yet somehow.

galmiklos, 2025-02-11:
Instead of just brutally using a hard coded number as timeout, I think I found the correct solution.

I modified base.py, so it looks like this.

```
class ScrapeSupplier(Supplier):
    session: Session
    cookies = CookieJar()

    extra_headers = {}
    fallback_domains = [None]

    request_timeout = 0.0
    retry_timeout = 0.0

    if config := get_config():
        request_timeout = config["request_timeout"]
        retry_timeout = config["retry_timeout"]

    def scrape(self, url) -> Response | None:
        if not hasattr(self, "session"):
            self._setup_session()

#        if config := get_config():
#            self.request_timeout = config["request_timeout"]
#            self.retry_timeout = config["retry_timeout"]

        for retry in retry_timeouts():
```

30350n, 2025-02-11:
Good catch, fixed in [1.8.1](https://github.com/30350n/inventree_part_import/releases/tag/1.8.1).