30350n/inventree-part-import #63: AttributeError: 'NoneType' object has no attribute 'exact_manufacturer_products_count'

zmdev2082, 2025-02-27:
I'm getting the following error when trying to run the tool for the first time with various digikey parts. Below is the output with verbose turned on.

`$ inventree-part-import 1597-114090050-ND --verbose

setting up InvenTree API ...
loading api configuration from 'inventree.yaml' ...
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/ HTTP/1.1" 200 475
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/user/me/ HTTP/1.1" 200 133
(only 1 of 5 available supplier modules are configured)
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/company/18/ HTTP/1.1" 200 340
setting up categories ...
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/part/category/ HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/part/parameter/template/ HTTP/1.1" 200 422
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/part/category/parameters/ HTTP/1.1" 200 1365
successfully setup categories!

DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 10.87.13.150:80
DEBUG:urllib3.connectionpool:http://10.87.13.150:80 "GET /api/part/parameter/template/ HTTP/1.1" 200 422
searching for 1597-114090050-ND ...
searching at DigiKey ...
DEBUG:digikey.oauth.oauth2:Using API V3
DEBUG:digikey.oauth.oauth2:Using API V3
Traceback (most recent call last):
  File "/home/humble/.local/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/cli.py", line 23, in wrapper
    func(*args, **kwargs)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/cli.py", line 210, in inventree_part_import
    importer.import_part(part, None, supplier, only_supplier)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 64, in import_part
    results, result_count = async_results.get()
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 774, in get
    raise self._value
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/suppliers/base.py", line 95, in cached_search
    return self.search(search_term)
  File "/home/humble/.local/pipx/venvs/inventree-part-import/lib/python3.10/site-packages/inventree_part_import/suppliers/supplier_digikey.py", line 59, in search
    if results.exact_manufacturer_products_count > 0:
AttributeError: 'NoneType' object has no attribute 'exact_manufacturer_products_count'`

zmdev2082, 2025-02-27:
I just configured mouser, and I got the exact same error. Based on the logs it seemed like it found the correct part before breaking.

`$ inventree-part-import -s mouser 511-STM32F3DISCOVERY
(only 2 of 5 available supplier modules are configured)
successfully setup categories!

searching for 511-STM32F3DISCOVERY ...
importing Mouser part 511-STM32F3DISCOVERY ...
error: failed to match category for 'Embedded Solutions / Engineering Tools / Embedded Processor Development Kits / Development Boards & Kits - ARM'
Traceback (most recent call last):
  File "/home/humble/.local/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
`

30350n, 2025-02-27:
Hmm that one looks weird, is that reproducible for you?

zmdev2082, 2025-02-27:
Since I only just signed up the response I was getting the v3 api was returning none.

I used this pull request [https://github.com/peeter123/digikey-api/pull/51] and had to change some things to make it actually compatible with python 3.12.

Then I have updated suppliers_digikey.py to work with the v4 api. Seems to be working now.

zmdev2082, 2025-03-02:
You can no longer sign up for v3.

30350n, 2025-03-02:
Ah, well that's a problem I guess ...

mawoka-myblock, 2025-04-05:
Same here. As the DigiKey python library seems unmaintained, one may have to look for a replacement

zmdev2082, 2025-04-06:
@mawoka-myblock I submitted a pull request that uses v4 #64. You can pull my fork and use it just note the issues with pricing that I mentioned. Haven't heard anything from @30350n...

zmdev2082, 2025-04-06:
> Same here. As the DigiKey python library seems unmaintained, one may have to look for a replacement

@mawoka-myblock I submitted a pull request that uses v4 #64. You can pull my fork and use it just note the issues with pricing that I mentioned. Haven't heard anything from @30350n...

zmdev2082, 2025-04-07:
@30350n I'm sure you've read https://github.com/peeter123/digikey-api/pull/51

Ki-nTree are using https://github.com/hurricaneJoef/digikey-api/tree/master which I started to use originally but since I liked how 
https://github.com/flaviut/digikey-api/tree/master implemented the usage of the API I used that.  I think I also had some issues with python 3.12.

I've been meaning to look at how to incorporate a second call to [ProductPricing](https://developer.digikey.com/products/product-information-v4/productsearch/productpricing) to get the price break data.

Might be easier to do that initially in `supplier_digikey.py` rather than forking https://github.com/flaviut/digikey-api/ with the hope that they eventually start merging pr's

closed 2025-10-04 by 30350n