30350n/inventree-part-import #65: Error getting Mouser details from URL

linguini1, 2025-03-06:
When running this command to add a part, I get the following output:
```
(env) inventree@inspace-server:~/inventree$ (env) inventree@inspace-server:~/inventree$ inventree-part-import --supplier mouser CRCW08052K20JNEA
/home/inventree/inventree/env/lib/python3.10/site-packages/thefuzz/fuzz.py:11: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
warning: failed to load cookies from browser 'none' (not in [chrome, chromium, opera, opera_gx, brave, edge, vivaldi, firefox, librewolf, safari, lynx, w3m, arc])
warning: failed to load cookies from browser 'none' (not in [chrome, chromium, opera, opera_gx, brave, edge, vivaldi, firefox, librewolf, safari, lynx, w3m, arc])
(only 1 of 5 available supplier modules are configured)
warning: category 'Electronics/Development Boards' on host is not defined in categories.yaml
successfully setup categories!

searching for CRCW08052K20JNEA ...
warning: failed to get parameters from 'https://www.mouser.ca/ProductDetail/Vishay-Dale/CRCW08052K20JNEA?qs=SM0pg8AlB3oVgqQGjPYk5Q%3D%3D' (might be blocked)
Traceback (most recent call last):
  File "/home/inventree/inventree/env/bin/inventree-part-import", line 8, in <module>
    sys.exit(inventree_part_import())
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/inventree_part_import/cli.py", line 23, in wrapper
    func(*args, **kwargs)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/inventree_part_import/cli.py", line 210, in inventree_part_import
    importer.import_part(part, None, supplier, only_supplier)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 86, in import_part
    import_result |= self.import_supplier_part(supplier, api_part, existing_part)
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/inventree_part_import/part_importer.py", line 220, in import_supplier_part
    url = self.api.base_url + supplier_part.url[1:]
  File "/home/inventree/inventree/env/lib/python3.10/site-packages/inventree/base.py", line 353, in __getattr__
    return super().__getattribute__(name)
AttributeError: 'SupplierPart' object has no attribute 'url'. Did you mean: '_url'?
```

30350n, 2025-03-07:
Seems like newer InvenTree API versions don't include a `url` field anymore in the reponse.

matmair, 2025-03-07:
we thought that no one was using them - see https://github.com/inventree/InvenTree/pull/9150 - if you need them we can re-add them for sure.
Note that this was only changed on latest

30350n, 2025-03-07:
Thanks, but this is fine actually. I can just manually construct the url from the `pk`. There's really no reason to include it in the API reponse, this is a good change 👍

30350n, 2025-03-07:
Eh, nvm that actually. The `URL`/`_url` fields in `inventree_python` actually refer to the API url of course, which is different from the web url. Interestingly the web url for platform UI is also different yet again:

```
API URL: <BASE>/company/part/<PK>
WEB URL: <BASE>/supplier-part/<PK>
PIO URL: <BASE>/platform/purchasing/supplier-part/<PK>
```

The way I use it is really no critical though, so I'm just going to hardcode it I guess. Would probably still be nice if this could be unified somehow? Not sure.