30350n/inventree-part-import #69: Cannot import more than one part

norandomtechie, 2025-05-19:
Hi, thank you for your work on this great tool!  I have an Inventree instance set up with Docker + nginx, and I wanted to use this plugin to import a bunch of parts from Digikey.  I got it working using the fork at https://github.com/zmdev2082/inventree_part_import/tree/merge-v4 but that repo doesn't have the Issues working, so I'm hoping this is just a simple issue that could be addressed here.

When I try running:
`inventree-part-import parts.csv` 

Where parts.csv contains some 74-series gates and the TDCR1050M seven-segment display.

I get the following:

```
importing 'parts.csv' as single column csv file
is the first row 'Parts' a header? (Y/N) 
(only 1 of 5 available supplier modules are configured)

setup default categories/parameters configuration:
> Copy categories from InvenTree
  Copy default categories configuration
  Create empty configuration (manual setup)
successfully setup categories!

searching for SN74HC00N ...
successfully updated DigiKey part 2156-SN74HC00N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC02N ...
successfully updated DigiKey part 2156-SN74HC02N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC04N ...
successfully updated DigiKey part 2156-SN74HC04N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC08N ...
successfully updated DigiKey part 2156-SN74HC08N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC10N ...
successfully updated DigiKey part 2156-SN74HC10N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC14N ...
successfully updated DigiKey part 296-1577-5-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for SN74HC74N ...
successfully updated DigiKey part 2156-SN74HC74N-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

searching for TDCR1050M ...
successfully updated DigiKey part TDCR1050M-ND (https://digisim2.ecn.purdue.edu/supplier-part/3/)

successfully imported all parts!
```

However, on Inventree, only the last part (TDCR1050M) shows up in the Parts list, as if the last updated part is actually overwriting the existing part instead of creating a new part.  Is there a setting I'm missing that creates new parts, or is this a bug in the fork/original code?

30350n, 2025-05-19:
Hmm, yeah for some reason all these Parts use the same SupplierPart.

Can you check if this also happens:

1. for a `csv` file containing **new** parts (only if this wasn't the first time you ran this for this `csv` file)
2. if you import the same parts by providing them via the command line (`inventree-part-import 2156-SN74HC00N-ND 2156-SN74HC02N-ND 2156-SN74HC04N-ND ...`)
3. happens for parts imported from other suppliers

norandomtechie, 2025-05-19:
Thanks for your time!  I tried the three things above, with clean-ups in the middle as follows:

1. Started from a fresh Inventree instance running on Docker through an nginx proxy.  Set up inventree-part-import from zmdev2082's fork in a Python virtual environment.
2. Created a CSV file with just 2156-SN74HC00N-ND and uploaded it via `inventree-part-import`.  The 74HC00 part was created on Inventree.
3. Changed the CSV file to **only** contain 2156-SN74HC02N-ND and uploaded it.  The 74HC00 part gets overwritten with 74HC02, including Supplier and Manufacturer Part.
4. I delete the 74HC02 part on Inventree so that there's no parts at all, and try `inventree-part-import 2156-SN74HC10N-ND 2156-SN74HC02N-ND 2156-SN74HC04N-ND`.  This is the log I'm getting:

```
❯ inventree-part-import 2156-SN74HC10N-ND 2156-SN74HC02N-ND 2156-SN74HC04N-ND
(only 1 of 5 available supplier modules are configured)
successfully setup categories!

searching for 2156-SN74HC10N-ND ...
successfully updated DigiKey part 2156-SN74HC10N-ND (https://digisim2.ecn.purdue.edu/supplier-part/6/)

searching for 2156-SN74HC02N-ND ...
successfully updated DigiKey part 2156-SN74HC02N-ND (https://digisim2.ecn.purdue.edu/supplier-part/6/)

searching for 2156-SN74HC04N-ND ...
successfully updated DigiKey part 2156-SN74HC04N-ND (https://digisim2.ecn.purdue.edu/supplier-part/6/)

successfully imported all parts!
```

5. I tried a completely different part, SC1510 (Raspberry Pi RP2350B microcontroller), and that still overwrites the previous part (which was 74HC04).

It is very weird that it's matching against a completely unrelated part.  I'll dive through the Python as well to see why this is the case.

30350n, 2025-05-19:
SupplierParts get retrieved here: https://github.com/30350n/inventree-part-import/blob/7237a3dee4406be1fe8d476f1c56d4680f8834a8/inventree_part_import/inventree_helpers.py#L23-L33

So it would be interesting what `sku` is set to in your case and whether or not the `SKU`s of the returned `supplier_parts` actually match that.

norandomtechie, 2025-05-19:
Thanks!  Okay, I first added the part SN74HC02N, whose SKU is 2156-SN74HC02N-ND.

Then, I added the part SC1510 while debugging the script.  At line 25:

1. The parameter `sku` in `SupplierPart.list(inventree_api, SKU=sku)` is 2648-SC1510TR-ND.
2. In the returned array from `SupplierPart.list`, `supplier_parts[0].SKU` is 2156-SN74HC02N-ND.

I think the Inventree API is just... broken?  There is no reason at all that a query for SKU `2648-SC1510TR-ND` (a microcontroller) should return `2156-SN74HC02N-ND` (a quad 2-input OR gate).

I'm going to add in new checks for the SKU and see what that does, but I think this might just need to be filed with Inventree's repo.

30350n, 2025-10-04:
Closing this as you seem to have been the only one ever experiencing this.

(Feel free to ask me to reopen this if its still relevant ^^)