30350n/inventree-part-import #70: Installing on Docker

bdmartens, 2025-06-22:
I've been trying to get this up and running on my docker Inventree 0.17.14
So far, additional complications require:
Console in via Portainer (or docker exec)

apk add build-base   (gcc not part of inventree docker image)
pip install pipx
pipx install inventree-part-import

then I'm still not sure if installing this way will persist.
Do I need to add anything to the plugins.txt?  What would I add there?

If this is correct, can you update your install instructions?
Thanks

30350n, 2025-06-22:
inventree-part-import offers a CLI which communicates with the API of your InvenTree instance. You are neither required nor encouraged to run it on the same machine (or container) your instance is running on, ideally you should run it on a machine ... that you can actually access easily.

>  Do I need to add anything to the plugins.txt? What would I add there?

It is not a plugin, so no.

bdmartens, 2025-06-22:
I'm having more issues with this install.
First run of inventree-part-import, it does not prompt a first setup (unlike an attempt I did earlier).
I get an error at the top of any time I run it such as:

```
inventree-part-import --help
error: failed to load supplier module 'supplier_digikey' with No module named 'six'
```

I don't seem to be able to run 
```
inventree-part-import --configure digikey
error: failed to load supplier module 'supplier_digikey' with No module named 'six'
Usage: inventree-part-import [OPTIONS] [INPUTS]...
Try 'inventree-part-import --help' for help.

Error: Invalid value for '--configure': 'digikey' is not one of 'tme', 'lcsc', 'mouser', 'reichelt'.
```
I got a lot of errors when trying to configure mouser, which led me to throw this at gemini to help out, given googlefails.
It appears there was the package 'six' needed, I hadn't been aware that was a package.

pip install six
pipx inject inventree-part-import six

I can then seem to access digikey configuration.
I am not sure if this is common, if it is something due to a convoluted setup on my part, due to version changes in inventree docker images, or something else entirely.

closed 2025-10-04 by 30350n