30350n/inventree-part-import #78: LCSC stopped working?

Zontex, 2025-09-14:
Today tried to work with it but I see cannot fetch product from LCSC, do you guys face the same issue?

Zontex, 2025-09-14:
seems search API something gone wrong. I bypassed it by:

'''
 def search(self, search_term):
        """
        Since we always pass an LCSC product code (e.g. C701346),
        skip the search API and query the product detail API directly.
        """
        code = search_term.strip().upper()

        if not (code.startswith("C") and code[1:].isdigit()):
            warning(f"LCSC: invalid product code format '{search_term}'")
            return [], 0

        url = PRODUCT_INFO_URL.format(code)
        r = self.scrape(url)

        if r and (result := r.json().get("result")):
            return [self.get_api_part(result)], 1

        warning(f"LCSC: failed to retrieve product detail for {code}")
        return [], 0
'''

G-Pereira, 2025-09-21:
Now I understood. You are looking up the LCSC part number directly

30350n, 2025-10-04:
Should be fixed with [`67261e0`](https://github.com/30350n/inventree-part-import/commit/67261e0c031718bc573045814abe0f98592f3f82).

closed 2025-10-04 by 30350n